cotestatnt/esp-fs-webserver #1: default site

trombose009, 2022-01-09:
Hi, 
is there a way to set the default website to something else than /edit?  For example to /index.html?

cotestatnt, 2022-01-09:
Hi @trombose009 
The default page handled on root address is [/setup](/setup) only if [/index.htm](/index.htm) is not present in filesystem.
Once you have uploaded the file, it will be opened on root address (or with the complete filename offcourse), but it search only for **.htm** extension and not with **.html**
But you are right, should be opened also if file is index.html, thanks for your notice.
I will add the handler for the next release!

cotestatnt, 2022-01-09:
Check this commit if you want fix code yourself:

https://github.com/cotestatnt/esp-fs-webserver/commit/3b8522586ef5c5197163ed17ac10dc4cf056351e

closed 2022-01-09 by trombose009