cotestatnt/esp-fs-webserver #3: edit page  shows FS INIT ERROR and not displaying Files

thornhilldenis, 2022-03-17:
Hi cotestatnt  , many thanks for this library very useful for every developer .
i Am developing  with  esp32  wroom dev   my small project .

I did install your ESP-FS-webserver   and it worked like a charm .   on my first install 
BUT suddenly something strange happened . ?

i have not change board ,partitions,  on my arduino ide keeps the same as when it worked like a charm .
The code i am uploading is yours .inos from EXAMPLES .

When it opens /edit   it shows everything  but the files , and also indicates  FS INIT ERROR .
i am using the  handleformdata  example .

So i did install simpleserver  and also the same  FS INIT ERROR
I did use other esp32 out of the package and the same FS INIT ERROR
I did replace all library with fresh download new ones   and the same . FS INIT ERROR
I test it with  mozzilla , safari , edge , many browsers just in case  but FS INIT ERROR

So i decided to upload  spiffs  test and runs ok , even lists my files meaning data files of your library .

i also uploaded de FS library   FSBROWSER that also use ACE and it does shows the files , but it hangs .

So i have being the last 3 days trying to solve this so i can continue with my development but i am stucked on this mysterious  situation that i cant not understand .

If you could be kind , and at least tell me where to go , check, replace, look , or any idea about my situation i will much appreciate ,

Many thanks in advance in case you have the time for an answer  to me ,

Denis Thornhill

cotestatnt, 2022-03-17:
Hi @thornhilldenis .
On which version of Arduino core are you working?
And withwhich kind of dev-board?

If you give me some more specific information, I can try to reproduce this strange issue with my test setup.

thornhilldenis, 2022-03-17:
many thanks for your response .
i am using  a chinesse nodemcu esp 32 dev board  38  pins 
Chip says   esp - wroom 32  

does not have a version written on the board  
The most strange as I explained to you is that your library   worked like a charm ,   and i even  uploaded , renamed , deleted,  files  perfectly  ..  

At this exact moment i am installing arduino ide  , your library  on another PC  , from zero ,  to verify if the problem could be on my original PC  i was working with  could be  corrupted  some how ..

I will let you know in approx 1 hour from now while i am installing all the new stuff

thornhilldenis, 2022-03-17:
upps sorry  , i am working with arduino  ide  1.8.13

thornhilldenis, 2022-03-17:
Arduino avr 1.8.4

thornhilldenis, 2022-03-17:
hi my friend .

Report for cotestatnt
i did mount all on a separate system from zero .
Other PC 
New install Arduino ide 1.8.19
Avr  1.8.5
I downloaded your library esp-fs   from git  into that other PC 
I did install  into tools ESP32FS-1.0  so I can upload data files 
I opened a new nodemcu esp 32 never used a new package .
I uploaded data to flash  ok 
I uploaded your esp-fs- from examples library into esp32  OK    
comenting lines  
 else {
    Serial.println("ERROR on mounting filesystem. It will be formatted!");
  /// FILESYSTEM.format();
  /// ESP.restart();
  }
  So to prevent it won't keep restarting and formating  in a loop 
  
  And the result is just the same as on the other PC station.
  Everything loads    ace shows   but   FS INIT ERROR 
  
  I rin Spiffs test,  formating false, just to check files are still there   >>>  and yes they are.
  
  FS  INIT ERROR
  
  Awaiting your comments or instructions.
  Sorry to bother you with this  .
  
  Many thanks in advance.
  I will like to get to the end of solving this problem .
  
  Thanks  again 
  Denis Thornhill

cotestatnt, 2022-03-18:
HI @thornhilldenis maybe I've undestood whats your behaviuor.
You say your board it's a NodeMCU like, so I presume you have selected this kind of board with Arduino IDE.
Unfortunately this type of card does not provide for the choice of the partition scheme but uses the SPIFFS by default.
ACE editor and Setup page it's embedded in firmware so it will always work.

The examples however are written using FFat as the filesystem type: SPIFFS is really obsolete in my opinion, FFat it's faster, LittleFS maybe it's even better, but it was only introduced in the latest versions of the ESP32 core (starting from V2.0.0 if I remember well).

You can choose a different type of board where you can set the partition scheme for using FFat (like ESP32-Dev module for example) or change the example in order to use SPIFFS

[image]

```
.....
#elif defined(ESP32)
  #include <SPIFFS.h>
  #define FILESYSTEM SPIFFS
#endif
```
If you choose to use FFat, the plugin you use for upload SPIFFS is no longer good, you have to install one that works with FFat or LittleFS (like this for example [arduino-esp32fs-plugin](https://github.com/lorol/arduino-esp32fs-plugin)) or you can simply use the ACE editor to upload all needed files directly with browser.

P.S.
I've edited all the examples of library in order to use LittleFS as default both for ESP8266 and ESP32 because it's not needed to set the partition scheme like with FFat (it works with default).

thornhilldenis, 2022-03-18:
Hello good morning !!
Dear Developer. cotestatnt

first of all, I want to thank you  for your time spent on my case,

Following your instructions, I want you to know that just uploading with partition selection as you describe to me in the image,  and uncommenting the two lines,  IT WORKED PERFECTLY AS YOU SAID .
WORKS  beautiful .

My esp32 MCU like, does works with board  esp32 dev just fine.

I also tested with a Chinese wemos  wifi&bluetooth battery OLED 0.96  and also works perfectly.

No more to say then, but thank you for the teaching class,   and all I have learned from your explanation about partitions and in love with your esp-fs library that I will be using on most of my projects .

This was my first time, on GitHub asking anything from someone, and  I am amazed and satisfied with your response and collaboration .

I want to know if should I close this issue or not, I don't know what to do about it as  I said new on Github.
Please let me know what to do ?

Thanks again.
Denis thornhill

thornhilldenis, 2022-03-19:
Well, thanks for your cooperation .
i will close this isue  
Best regards.

closed 2022-03-19 by thornhilldenis