cotestatnt/esp-fs-webserver #12: changing the code to automatically inlcude the correct  Little-filesystem library ESP8266 <LittleFS.h>  / ESP32)<LITTLEFS.h>

StefanL38, 2022-06-26:
Hi I made a small change to the code so that the correct littl-flsesystem-library is included on compiling
here is my modified version

I have tested it with an ESP32 and an ESP8266-board and it does compile and run as expected

the modification is here
`#ifdef ESP8266
#include <LittleFS.h>
#define FILESYSTEM LittleFS
ESP8266WebServer server(80);

#elif defined(ESP32)
#include <LITTLEFS.h>
#define FILESYSTEM LITTLEFS
WebServer server(80);
#endif

`

full code is this

```
#include <esp-fs-webserver.h>   // https://github.com/cotestatnt/esp-fs-webserver

#include <FS.h>

#ifndef LED_BUILTIN
#define LED_BUILTIN 2
#endif

// Test "options" values
bool boolVar = true;
uint32_t longVar = 1234567890;
String stringVar = "Test option String";
uint8_t ledPin = LED_BUILTIN;

int myTestVar1;
int myTestVar2;
int myTestVar3;

// Timezone definition to get properly time from NTP server
#define MYTZ "CET-1CEST,M3.5.0,M10.5.0/3"
struct tm Time;

#ifdef ESP8266
#include <LittleFS.h>
#define FILESYSTEM LittleFS
ESP8266WebServer server(80);

#elif defined(ESP32)
#include <LITTLEFS.h>
#define FILESYSTEM LITTLEFS
WebServer server(80);
#endif

FSWebServer myWebServer(FILESYSTEM, server);

////////////////////////////////  Filesystem  /////////////////////////////////////////
void startFilesystem() {
  Serial.println( F("(try to start FileSystem") );
  // FILESYSTEM INIT
  if ( FILESYSTEM.begin()) {
    Serial.println( F("FILESYSTEM.begin() done") );
    File root = FILESYSTEM.open("/", "r");
    File file = root.openNextFile();
    while (file) {
      const char* fileName = file.name();
      size_t fileSize = file.size();
      Serial.printf("FS File: %s, size: %lu\n", fileName, (long unsigned)fileSize);
      file = root.openNextFile();
    }
    Serial.println();
  }
  else {
    Serial.println( F("ERROR on mounting filesystem. It will be formmatted!") );
    FILESYSTEM.format();
    Serial.println( F("formatting done ESP.restart()") );
    ESP.restart();
  }
}

////////////////////  Load application options from filesystem  ////////////////////
bool loadApplicationConfig() {
  StaticJsonDocument<1024> doc;
  File file = FILESYSTEM.open("/config.json", "r");
  if (file) {
    Serial.println( F("successfully opened file config.json for read") );
    DeserializationError error = deserializeJson(doc, file);
    file.close();
    if (!error) {
      Serial.println(F("Deserializing config JSON.. successful"));
      boolVar = doc["A bool var"];
      stringVar = doc["A String var"].as<String>();
      longVar = doc["A long var"];
      ledPin = doc["LED Pin"];
      serializeJsonPretty(doc, Serial);
      Serial.println();
      return true;
    }
    else {
      Serial.println( F("Failed to deserialize JSON. File could be corrupted"));
      Serial.println(error.c_str());
    }
  }
  return false;
}

void setup() {
  Serial.begin(115200);
  Serial.println( F("Setup-Start") );
  // FILESYSTEM INIT
  startFilesystem();

  // Try to connect to flash stored SSID, start AP if fails after timeout
  IPAddress myIP = myWebServer.startWiFi(15000, "ESP8266_AP", "123456789" );

  // Load configuration (if not present, default will be created when webserver will start)
  if (loadApplicationConfig()) {
    Serial.println( F("Application option loaded") );
  }
  else {
    Serial.println( F("Application NOT loaded!") );
    Serial.print( F("Open http://"));
    Serial.print(myIP);
    Serial.println( F("/setup to configure parameters") );
  }

  // Configure / setup page and start Web Server
  myWebServer.addOption(FILESYSTEM, "LED Pin", ledPin);
  myWebServer.addOption(FILESYSTEM, "A long var", longVar);
  myWebServer.addOption(FILESYSTEM, "A String var", stringVar.c_str());
  myWebServer.addOption(FILESYSTEM, "A bool var", boolVar);

  myWebServer.addOption(FILESYSTEM, "my number 1", myTestVar1);
  myWebServer.addOption(FILESYSTEM, "this testvar2", myTestVar2);
  myWebServer.addOption(FILESYSTEM, "3 my 33 myTestVar3", myTestVar3);

  if (myWebServer.begin()) {
    Serial.print( F("ESP Web Server started on IP Address: ") );
    Serial.println(myIP);
    Serial.println();

    Serial.println( F("Open ") );
    Serial.print( F("http://") );
    Serial.print(myIP);
    Serial.println( F("/setup") );
    Serial.println( F("to configure optional parameters") );
    Serial.println();

    Serial.println( F("Open ") );
    Serial.print( F("http://") );
    Serial.print(myIP);
    Serial.println( F("/edit") );
    Serial.println( F("to view and edit files") );
    Serial.println();

    Serial.println( F("Open ") );
    Serial.print( F("http://") );
    Serial.print(myIP);
    Serial.println( F("/update") );
    Serial.println( F("to upload firmware and filesystem updates"));
    Serial.println();
    Serial.println( F("start looping myWebServer.run()") );
  }
}

void loop() {
  myWebServer.run();
}

```

StefanL38, 2022-06-29:
> Probably you are using an old version of core (latest is 2.0.3).

2.0.3 doesn't make sense to me. Is this a type and you mean ESP8266 3.0.2

`2.0.3
3.0.2`

which version do you mean ESP32 my Boardmanager tells me 
newest version for ESP32 is 
[image]

newest version for ESP8266 is
[image]

cotestatnt, 2022-06-29:
You are using an old JSON link in your IDE preferences for ESP32. 
Probably you've got it from an outdated tutorial instead from Espressif official documentation.
You can find the updated information for [Arduino IDE at this address](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html)

[image]

[image]

closed 2022-07-07 by cotestatnt