cotestatnt/esp-fs-webserver #14: ESP8266 / ESP32 webserver ace js tryit alike live preview

ldijkman, 2022-07-10:
hello

i am using ace on ESP8266 / ESP32 webserver
for wifi relais switches
i would like to ad live html preview to the ace code text editor (tryit codepen jsfiddle alike)

like https://github.com/geard-dev/web-based-live-code-editor
demo https://live-editor.js.org/html.html

but i do not now how to implement it in my editor code 

is there anyone that can help me?

Greet Luberth
https://github.com/ldijkman/randomnerd_esp32_wifi_manager/tree/main/LAB_Experiments

```
<!DOCTYPE html>
<html lang="en">
<meta http-equiv="Content-type" content="text/html; charset=utf-8">
<title>Electra's ESP Editor</title>
<link rel="apple-touch-icon" href="/ace.ico" type="image/x-icon">
<link rel="shortcut icon" href="/ace.ico" type="image/x-icon">
<link rel="icon" href="/ace.ico" type="image/x-icon">
<style type="text/css" media="screen">
  label {
    font-size: 12px;
    font-family: sans-serif
  }

  .cm {
    z-index: 300;
    position: absolute;
    left: 5px;
    border: 1px solid #444;
    background-color: #f5f5f5;
    display: none;
    box-shadow: 0 0 10px rgba(0, 0, 0, .4);
    font-size: 12px;
    font-family: sans-serif;
    font-weight: 700
  }

  .cm ul {
    list-style: none;
    top: 0;
    left: 0;
    margin: 0;
    padding: 0
  }

  .cm li {
    position: relative;
    min-width: 60px;
    cursor: pointer
  }

  .cm span {
    color: #444;
    display: inline-block;
    padding: 6px
  }

  .cm li:hover {
    background: #444
  }

  .cm li:hover span {
    color: #eee
  }
[... 467 lines cut ...]
<script id="ace" src="/acefull.js" charset="utf-8"></script>
<script>
  if (void 0 === ace.edit) {
    var script = document.createElement("script");
    script.src = "/ace.js", script.async = !1, document.head.appendChild(script)
  }
</script>

<body onload="onBodyLoad()">
  <div id="loader" class="loader"></div>
  <div id="uploader"></div>
  <div id="tree"></div>
  <div id="editor"></div>
  <div id="preview" style="display:none"></div><iframe id="download-frame" style="display:none"></iframe>

```
above code results in next screenshot (running on ESP32 webserver http://garage.local/edit)

[image]

would like to have a live html preview panel added
like demo on https://github.com/geard-dev/web-based-live-code-editor
demo https://live-editor.js.org/html.html

[image]

cotestatnt, 2022-07-10:
Ho @ldijkman 
This library already has ACE editor included in source. You can use with <your_device_ip>/edit webpage in order to edit yhe source files directly on the esp filesystem.

Regarding the preview, this feature is not avalaible but in my opinion is also not necessary because you can simply open another browser tab, edit whatever you want in your own webpage and imnediatly check the result reloading page.

ldijkman, 2022-07-10:
yes your right
i can place two browser windows side by side
one with editor and one with html preview

but i think it would be nice like next codepen
https://codepen.io/ldijkman/full/LYdZpYp

cotestatnt, 2022-07-11:
A preview box like that, I think is certainly useful for quickly trying small snippets of code (which is why there are services like Codepen or Jsfiddle, and not entire web pages.
Personally I still think it is useless, however you can simply put the files of the repository you have linked into the filesystem of the esp if you like.

If you want to try entire websites I recommend using a service like CodeSandbox ([this is your example](https://codesandbox.io/s/beautiful-shadow-82rysi)) where you can put all the files needed.

This library has the main objective of serve the web pages you upload to the filesystem, this means that the development of these web pages can be done anywhere: online, with your local webserver running on a PC or directly with the ESP.

closed 2022-07-14 by cotestatnt