cotestatnt/esp-fs-webserver #18: add multiple OptionBox

Blueforcer, 2023-02-19:
Hello,

i created a with addOptionBox like in the optionsExample.
is it possible to add multiple option sites with different settings?

cotestatnt, 2023-02-20:
Hi @Blueforcer 
I'm not sure if I understand your question.
Perhaps you intend to change the CSS styling of each section, or is it something else? Could you possibly give an example?

With the new features you can add as many "option boxes" as you want.
In the latest release I also added the possibility of injecting custom CSS, which allows you to define the style of both the new elements added, but also to override the style of what is already used by default.

Check the updated example [customHTML.ino](https://github.com/cotestatnt/esp-fs-webserver/tree/master/examples/customHTML) for more details.

[image]

In any case, if you want to radically change the look of the page, the advice is to directly modify the HTML/CSS/Js sources that you find [in this folder](https://github.com/cotestatnt/esp-fs-webserver/tree/master/setup-ui/data) and then overwrite the setup_htm.h file with the one obtained by following the instructions in the [readme](https://github.com/cotestatnt/esp-fs-webserver/blob/master/setup-ui/data/build/readme.md). inside the build folder.

Blueforcer, 2023-02-20:
i will try to explane:
 i made this option page
 
[image]

only with this few lines of code
 `mws.addOptionBox("MQTT");
  mws.addOption("Broker", mqtthost);
  mws.addOption("Port", mqttport);
  mws.addOption("Username", mqttuser);
  mws.addOption("Password", mqttpass);
  mws.addOption("Prefix", mqttprefix);
  mws.addOption("raw-html-button", save_btn_htm);
  mws.addJavascript(button_script);
`

so my question is, if it is possible to create a second optionpage in the same way (via code) or do i need to create custom html pages?

cotestatnt, 2023-02-20:
No, it's really simple indeed.
Every time you call the method `addOptionBox("label")`, the JavaScript "under the cover" create a new section and every option you add after will be showed in that new option box.

Keep the addJavascrit() or addCSS() at the end just for a clearer config.json file.

Blueforcer, 2023-02-20:
nice, i will try. thank you very much for your help and of course for your good work.
Im pretty sure you will see me again:)
Do you have a discord server or something like this? Or should i open a issue again for next questions?

closed 2023-02-20 by Blueforcer

Blueforcer, 2023-02-22:
Just a small hint:
if your add or rearange your options afterwards again the pages are completely messed up, you need to delete the config.json and reboot to have a correct config.json.
A function would be nice, wich rearange the config.json at boot to fit the addOptions sequence

cotestatnt, 2023-02-22:
Thanks for your hint, I will take in account for the next developments.

arimukhlas, 2023-04-03:
hi @cotestatnt  thanks for making this library, its great!!

i have question, 
1. How to show value of variable in custom HTML??
2. How do i clear the input, and go back to default
3. It will be better if you add label on Wifisetup page to give information about esp is connected or not, and when its connected, label also show local IP address. 
Thanks, youre great man!!