cotestatnt/esp-fs-webserver #19: Bool options are not saved

Blueforcer, 2023-02-20:
bool options, wich will be displayed as switches 
[image]

are not saved into the config.json. after i hit "save configuration" i check the config.json and the value is always false

cotestatnt, 2023-02-20:
You are right @Blueforcer 

I'm sorry, but this is a bug introduced with new setup UI because I have accidentally deleted a row of code.
I will fix it immediatly and then I will draft a new release.

cotestatnt, 2023-02-20:
Hi @Blueforcer 
I've just pushed the bugfix and some other functions for the /setup page (upload you custom logo file just clicking on default Espressif logo).

I want to do some more tests before drafting a new release, but if you want to fix this bug just overwrite the setup_htm.h file with the new one uploaded to Github.

Blueforcer, 2023-02-20:
youre pretty fast man! i think i can wait until next release.
Btw is it possible to get a dropdown as an option? :-)

cotestatnt, 2023-02-20:
> Btw is it possible to get a dropdown as an option? :-)

This is a feature I had already thought about, but it needs some work because the list can be of different data types.
I'll probably implement a version that allows you to use only strings or at most strings and numbers as the return value.

Blueforcer, 2023-02-20:
Strings are fine, or maybe just saving the selected index? Then the user can compare whatever he likes.

cotestatnt, 2023-02-21:
Goodmorning @Blueforcer 
The new release contains bug fix and the new method for dropdown listbox (only string values for the moment).

https://github.com/cotestatnt/esp-fs-webserver/releases/tag/1.2.0

Blueforcer, 2023-02-21:
Oh boy, that was damn fast! Nice work!

closed 2023-02-21 by Blueforcer

cotestatnt, 2023-02-21:
It was easier than i thought :)