cotestatnt/esp-fs-webserver #20: customOptions error

invidia191, 2023-02-22:
Hello,

I can not run the customOptions example on wemos d1 mini.

This error is in serial monitor:

02:54:25.747 -> 
02:54:25.747 -> AP mode.
02:54:25.747 -> Server IP address: 192.168.4.1
02:54:25.747 -> 
02:54:25.794 -> 
02:54:25.794 -> This are the current values stored: 
02:54:25.794 -> 
02:54:25.794 -> LED pin value: 2
02:54:25.794 -> Bool value: true
02:54:25.794 -> Long value: 1234567890
02:54:25.794 -> Float value: 15.500
02:54:25.794 -> String value: Test option String
02:54:25.794 -> Dropdown selected value: Monday
02:54:25.794 -> 
02:54:25.794 -> Application option loaded
02:54:25.837 -> 
02:54:25.837 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
02:54:25.837 -> 
02:54:25.837 -> Exception (3):
02:54:25.837 -> epc1=0x4020a743 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025145e depc=0x00000000
02:54:25.837 -> 
02:54:25.837 -> >>>stack>>>
02:54:25.837 -> 
02:54:25.837 -> ctx: cont
02:54:25.837 -> sp: 3ffffc90 end: 3fffffd0 offset: 0190
02:54:25.837 -> 3ffffe20:  00000000 000003e8 4021241c 00000000  
02:54:25.837 -> 3ffffe30:  00000000 00000000 3fff10cc 00000007  
02:54:25.837 -> 3ffffe40:  3ffffe80 00000009 00000020 40100b50  
02:54:25.837 -> 3ffffe50:  00000010 00000020 3ffffee0 3ffffed0  
02:54:25.837 -> 3ffffe60:  3ffeef40 c7efffff e54daff8 40209ce4  
02:54:25.837 -> 3ffffe70:  00000000 00000000 00000000 00000007  
02:54:25.883 -> 3ffffe80:  00000010 00000001 3ffffee0 40216133  
02:54:25.883 -> 3ffffe90:  e54daff8 00000001 3ffffee0 4021645c  
02:54:25.883 -> 3ffffea0:  3fff005c 3ffe8a45 3fff005c 40100b1c  
02:54:25.883 -> 3ffffeb0:  3ffe8a45 00000009 3ffffee0 3ffffed0  
02:54:25.883 -> 3ffffec0:  3ffeef40 c7efffff e54daff8 40209cd1  
02:54:25.883 -> 3ffffed0:  e54daff8 47efffff 00000000 3ff00000  
02:54:25.883 -> 3ffffee0:  3fff005c 0010001f 8900002d 3fff11ba  
02:54:25.883 -> 3ffffef0:  3fff178c 3fff18cc feefef00 feefeffe  
02:54:25.883 -> 3fffff00:  3fff18bc 3fff181c feefef20 feefeffe  
02:54:25.883 -> 3fffff10:  4021a4ac 00000000 0064616f 00000000  
02:54:25.930 -> 3fffff20:  feefeffe feefeffe feefeffe feefeffe  
02:54:25.930 -> 3fffff30:  feefeffe feefeffe feefeffe 4021a8b0  
02:54:25.930 -> 3fffff40:  00000000 000003e8 00000000 00000000  
02:54:25.930 -> 3fffff50:  00000000 00000000 402126f4 00000000  
02:54:25.930 -> 3fffff60:  00000000 3ffef16c 00000000 0010001f  
02:54:25.930 -> 3fffff70:  00000000 00000000 00000000 feefeffe  
02:54:25.930 -> 3fffff80:  feefeffe 4021a97c 0104a8c0 feefeffe  
02:54:25.930 -> 3fffff90:  3ffffed0 3ffffe80 feefeffe feefeffe  
02:54:25.930 -> 3fffffa0:  feefeffe feefeffe feefeffe 3ffef374  
02:54:25.930 -> 3fffffb0:  3fffdad0 00000000 3ffef348 40217790  
02:54:25.978 -> 3fffffc0:  feefeffe feefeffe 3fffdab0 40100df5  
02:54:25.978 -> <<<stack<<<
02:54:25.978 -> 
02:54:25.978 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
02:54:25.978 -> 
02:54:25.978 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,4)
02:54:25.978 -> 
02:54:25.978 -> load 0x4010f000, len 3424, room 16 
02:54:25.978 -> tail 0
02:54:25.978 -> chksum 0x2e
02:54:25.978 -> load 0x3fff20b8, len 40, room 8 
02:54:25.978 -> tail 0
02:54:25.978 -> chksum 0x2b
02:54:25.978 -> csum 0x2b
02:54:25.978 -> v000607e0
02:54:25.978 -> ~ld

Thx for the help.

cotestatnt, 2023-02-22:
You are right @invidia191, latest releases are bugged with ESP8266.

Unfortunally i've tried only to compile and not running, my mistake.

I will fix as soon as possibile

cotestatnt, 2023-02-22:
Hi @invidia191 
I've just pushed the sources that fix the bug you reported.

I will do further tests before creating a new release. 
If you want to try it yourself just download the repository as a zip archive and [install it manually](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries#importing-a-zip-library) using the Arduino IDE or you can also overwrite the actual files.

invidia191, 2023-02-23:
Working perfect, thanks.

But don't understand how to read saved variables and use them in loop.

cotestatnt, 2023-02-23:
@invidia191 in the customOptions.ino the value of variables are loaded at start-up calling the function [loadOptions()](https://github.com/cotestatnt/esp-fs-webserver/blob/master/examples/customOptions/customOptions.ino#L149)

For user convenience, the same thing is done when user do a request via the web interface because a [GET handler was defined](https://github.com/cotestatnt/esp-fs-webserver/blob/master/examples/customOptions/customOptions.ino#L155) for this purpose dinamically added at runtime as HTML/Javascript option.

As you can see in the source the `loadOptions()` function use the class method `myWebServer.getOptionValue(VAR_LABEL, variable);` for each variable you need to load at begin (configurations, tokens, addresses etc etc).

The equivalent method `myWebServer.saveOptionValue(VAR_LABEL, variable);` can be used to store permanently the value of variable in the config.json file (saved inside flash memory).

Of course this is just an example of how it is possible to use these methods which will then have to be adapted on a case-by-case basis.

However, keep in mind that a flash memory shouldn't be used to save "working variables" which change continuously due to the limit on the number of maximum writes (usually at least 100000)

invidia191, 2023-02-23:
That is easy. Thought needed arduinojson.
Great work !

closed 2023-03-13 by cotestatnt