cotestatnt/esp-fs-webserver #21: Special Characters in WiFi PW

Blueforcer, 2023-02-22:
It seems that special characters in wifi password are not saved correctly.
i have a "&-" at the end of my password.

If i print out my PW in console
https://github.com/cotestatnt/esp-fs-webserver/blob/4022ff91751d7ea3f218fccfa80a2e408c2ba93f/src/esp-fs-webserver.cpp#L158
its not there anymore.

It looks like the saving function has maybe an encoding problem

cotestatnt, 2023-02-22:
Hi @Blueforcer and thanks for your advice.
I need to to investigate deeper

Blueforcer, 2023-02-22:
Just for research purposes:
each password will be cut at an **&**
e.g.
**3EZeF!&tU_rB0-6vy8$T7ghlS5P1&-**
only **3EZeF!** is left

cotestatnt, 2023-02-22:
>Just for research purposes:

This is a very useful information indeed...
I suspect the & character causes the POST request handler to interpret the following as if it were a new parameter passed in the request body.

I have to change the way ssid and password are passed to the microcontroller (if possible also avoiding that the password is visible in clear text)

closed 2023-02-27 by Blueforcer