cotestatnt/esp-fs-webserver #23: Reducing filesystem access

Blueforcer, 2023-02-27:
[FEATURE REQUEST]

I love your lib, and i implement much options with it.
But due the ammount of options the access to the filesystem is horrible.
With all addOption and getOptionValue i get a total of 85 times opening the config.json file. Wich cause in a boottime of arround 25 seconds on a ESP32 with maximum clock of 240mhz.

is it possible to open the file and create the JsonDocument once?

cotestatnt, 2023-02-27:
Thank you @Blueforcer 
Your comments are very valuable and help me to develop the library in the right direction.

Regarding the methods you mentioned, I added them to make things easier for newbie users, but you can certainly avoid it and load/save the options all together by directly opening the config.json file using the ArduinoJson library.

As for the addOption() method instead, for now it's still necessary to open the file once for option (in read mode only if option is already present), but your comment gave me an idea that I'll work on as soon as possible.

cotestatnt, 2023-03-03:
Hi @Blueforcer 
I've just uploaded an updated [customHTML.ino example](https://github.com/cotestatnt/esp-fs-webserver/blob/master/examples/customHTML/customHTML.ino#L92) where options are loaded (and optionally saved) all in once using ArduinoJson library.

Unfortunately I haven't had the chance to work on the addOption() method yet.

Up now, every time the method is called, the _config.json_ file is read and only if the option is not already present is it also written.
In my tests the library seems fast enough even with very large _config.json_ file (5/10 kbytes), but avoiding opening the file if not neeed it's definitely a better option.

cotestatnt, 2024-01-19:
Hi @Blueforcer.
I know the answer is very late, but finally in the latest version I have drastically reduced the number of accesses to the configuration file (only once).

closed 2024-01-19 by cotestatnt