cotestatnt/esp-fs-webserver #24: ACE browser source

Blueforcer, 2023-03-17:
Is it possible to get the ACE file browser source?
I need to change something in it.

Blueforcer, 2023-03-17:
Well i found it :)
https://github.com/cotestatnt/esp-fs-webserver/tree/master/setup-ui/extras

will this also minified with node?

cotestatnt, 2023-03-17:
Hi Stephan.
The ACE editor came as is from the [FSBrowser.ino](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/FSBrowser/extras) example included in ESP8266 Arduino core

You can try to edit, but it's very hard because it's  mostly minified javascript and is very hard to interpret due to the meaningless function and variable names.

The free space info missing it's a well known bug, but it's due to [the C++ sources](https://github.com/cotestatnt/esp-fs-webserver/blob/master/src/esp-fs-webserver.cpp#L986) due to the different methods used in the filesystem management libraries with ESP32
LittleFS for example has the `usedBytes()` method while SPIFFS uses the `freeBytes()` method.

I promised myself I'd find a solution, but I didn't spend any more time on it.

Blueforcer, 2023-03-17:
> The free space info missing it's a well known bug, but came from [the C++ sources](https://github.com/cotestatnt/esp-fs-webserver/blob/master/src/esp-fs-webserver.cpp#L986) due to the different methods used in the filesystem management libraries with ESP32 LittleFS for example has the `usedBytes()` method while SPIFFS uses the `freeBytes()` method.
> 

thanks for this hint. since im using littleFS, i just replaced #include <FS.h> with <LittleFS.h>

#elif defined(ESP32)  
    totalBytes = LittleFS.totalBytes();  
    usedBytes = LittleFS.usedBytes();  
#endif

[image]

closed 2023-03-26 by Blueforcer