cotestatnt/esp-fs-webserver #26: Error run node minify.js to generate setup_htm.h

arimukhlas, 2023-04-06:
Hi @cotestatnt 

I have followed the instructions on [this link ](https://github.com/cotestatnt/esp-fs-webserver/issues/18#issuecomment-1436938221) 
- i put all files (app.js, setup.htm, and style.css) in folder build
- open a terminal in build folder
- run npm i
- run node minify.js

but I am getting an error 

[image]

- i also run npm install zlib

but the error still exist, can you give a hint?

cotestatnt, 2023-04-06:
Hi @arimukhlas 

I've done some edits still to be released. Yoi need to download the latest commits as zip file
[image]

The instructions needs to be updated. Once you have installed the latest avalaible zip you need to:
- open a terminal and set build folder as current. For example: 
 `cd <your_arduino_libraries_folder>esp-fs-webserver-master\setup-ui\data\build`
- run `npm i` to install script dependencies
- run `node minify.js`

The node.js script will generate a file named `setup_htm.h` inside build folder itself. Copy & paste this file in the `<your_arduino_libraries_folder>esp-fs-webserver-master\src` folder

arimukhlas, 2023-04-17:
@cotestatnt 

[image]

I have follow [your instruction](https://github.com/cotestatnt/esp-fs-webserver/issues/26#issuecomment-1498923297), but error still occur and setup_htm.h isnt generated, do you have any suggestion?

closed 2023-04-17 by arimukhlas