cotestatnt/esp-fs-webserver #28: How to update variable value on custom html example?

arimukhlas, 2023-07-25:
Hi @cotestatnt 

1. I implement your sugestion on [this](https://github.com/cotestatnt/esp-fs-webserver/issues/25#issuecomment-1498841906) and its work perfectly, so I try to other value like sensor DHT22. I want to show its value on custom_html, and the value update every 5 second without refresh webpage. I try to add some code to HTML

```
setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
    if (this.readyState == 4 && this.status == 200) {
      document.getElementById("label1").innerHTML = this.responseText;
      }
    };
    xhttp.open("GET", "/temperature", true);
    xhttp.send();
}, 5000 ) ;
```
but fail. Can you give me a hint to do that??

2. How to erase input previous settings every time I upload sketch?
3. When I try examples/simpleServer >> upload littleFS data, and then I try examples/customHTML, the previous HTML simpleServer still exist. How to reset it?

Thanks

cotestatnt, 2023-07-25:
Hi @arimukhlas 

Your piece of code has no particular problems. I tested it quickly on a test web page and it does what it's supposed to: it sends a `GET /temperature` request every 5 seconds.

1 - Are you sure you have implemented the response to the request on the "web server" side

2, 3 - When you ulpoad a new sketch, set the option _"Erase All Flash Before Sketch Upload"_ to **enabled**

arimukhlas, 2023-07-25:
> Hi @arimukhlas
>
> Your piece of code has no particular problems. I tested it quickly on a test web page and it does what it's supposed to: it sends a `GET /temperature` request every 5 seconds.
>
> 1 - [...]

1. I'm not sure, i just add code
```
myWebServer.addHandler("/temperature", HTTP_GET, handleTemp);
```

and function
```
void handleTemp() {
  WebServerClass* webRequest = myWebServer.getRequest();

  String reply = "45";
  webRequest->send(200, "text/plain", reply);
}
```

is that wrong? or what should i add in the code?

thanks

cotestatnt, 2023-07-25:
It should work.
What happens if you open the dev tools on browser (F12)? 
Check the log section, maybe there is some error which block the Javascript code execution

arimukhlas, 2023-07-25:
no error,

this is HTML
```
static const char custom_html[] PROGMEM = R"EOF(
<main class="container">
      <article class="grid">
        <div>
          <hgroup>
            <h1>ESP FS WebServer - LED Switcher</h1>
          </hgroup>
          <label for="remember">
            <input type="checkbox" role="switch" id="toggle-led" name="toggle-led">
            Toggle built-in LED
          </label>
          <br>
          <p id="esp-response"></p>
        </div>
      </article>
    </main>
							
<p id="label1"></p>

)EOF";

static const char custom_css[] PROGMEM = R"EOF(
pre{
    font-family: Monaco,Menlo,Consolas,'Courier New',monospace;
    color: #333;
    line-height: 20px;
    background-color: #f5f5f5;
    border: 1px solid rgba(0,0,0,0.15);
    border-radius: 6px;
    overflow-y: scroll;
    min-height: 350px;
    font-size: 85%;
}
.select{
  width: 25%;
  height:40px;
  padding-top: 10px;
  padding-left: 20px;
  border:1px solid #ccc;
  border-radius: 6px;
  box-shadow: 0 1px 2px 0 rgba(220, 220, 230, 0.5);
}
.body{
  background-color: cadetblue;
}
)EOF";

static const char custom_script[] PROGMEM = R"EOF(
function toggleLed() {
        const pars = new URLSearchParams({
          val:  document.getElementById('toggle-led').checked ? '1' : '0'
        });
        
        fetch('/led?' + pars )                // Do the request
        .then(response => response.text())    // Parse the response 
        .then(text => {                       // DO something with response
          console.log(text);
          document.getElementById('esp-response').innerHTML = text + ' <i>(Builtin LED is ON with a low signal)</i>';
        });
      }
      
      // Add event listener to the LED checkbox (the function will be called on every change)
      document.getElementById('toggle-led').addEventListener('change', toggleLed );
	  
	  setInterval(function ( ) {
				var xhttp = new XMLHttpRequest();
[... 228 lines cut ...]

void loop() {
  myWebServer.run();
}
void handleLed() {
  WebServerClass* webRequest = myWebServer.getRequest();

  // http://xxx.xxx.xxx.xxx/led?val=1
  if(webRequest->hasArg("val")) {
    int value = webRequest->arg("val").toInt();
    digitalWrite(ledPin, value);
  }

  String reply = "LED is now ";
  reply += digitalRead(ledPin) ? "OFF" : "ON";
  webRequest->send(200, "text/plain", reply);
}

void handleTemp() {
  WebServerClass* webRequest = myWebServer.getRequest();

  String reply = "45";
  webRequest->send(200, "text/plain", reply);
}
```

cotestatnt, 2023-07-25:
@arimukhlas I've tried your sketch and it works...

Are you sure about flash erase on upload? If some wrong configuration file is still in flash, it can cause problems

[image]

arimukhlas, 2023-07-25:
> @arimukhlas I've tried your sketch and it works...
> 
> Are you sure about flash erase on upload? If some wrong configuration file is still in flash, it can cause problems
> 
> [image]

Oh, I'm sorry...
when I implement on examples/simpleServer, its work perfectly... I can show value, 
but when I implement on examples/customHTML, it can not show value, I dont know why...

are you use my code above?

cotestatnt, 2023-07-25:
Yes of course.

Paste & copy directly from your post (all in one tab) and then uploaded on a ESP32 dev board erasing all the flash content.

```cpp
static const char custom_html[] PROGMEM = R"EOF(
<main class="container">
      <article class="grid">
        <div>
          <hgroup>
            <h1>ESP FS WebServer - LED Switcher</h1>
          </hgroup>
          <label for="remember">
            <input type="checkbox" role="switch" id="toggle-led" name="toggle-led">
            Toggle built-in LED
          </label>
          <br>
          <p id="esp-response"></p>
        </div>
      </article>
    </main>
<p id="label1"></p>
)EOF";

static const char custom_css[] PROGMEM = R"EOF(
pre{
    font-family: Monaco,Menlo,Consolas,'Courier New',monospace;
    color: #333;
    line-height: 20px;
    background-color: #f5f5f5;
    border: 1px solid rgba(0,0,0,0.15);
    border-radius: 6px;
    overflow-y: scroll;
    min-height: 350px;
    font-size: 85%;
}
.select{
  width: 25%;
  height:40px;
  padding-top: 10px;
  padding-left: 20px;
  border:1px solid #ccc;
  border-radius: 6px;
  box-shadow: 0 1px 2px 0 rgba(220, 220, 230, 0.5);
}
.body{
  background-color: cadetblue;
}
)EOF";

static const char custom_script[] PROGMEM = R"EOF(
function toggleLed() {
  const pars = new URLSearchParams({
    val:  document.getElementById('toggle-led').checked ? '1' : '0'
  });
  
  fetch('/led?' + pars )                // Do the request
  .then(response => response.text())    // Parse the response 
  .then(text => {                       // DO something with response
    console.log(text);
    document.getElementById('esp-response').innerHTML = text + ' <i>(Builtin LED is ON with a low signal)</i>';
  });
}

// Add event listener to the LED checkbox (the function will be called on every change)
document.getElementById('toggle-led').addEventListener('change', toggleLed );

setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
[... 183 lines cut ...]

void loop() {
  myWebServer.run();
}
void handleLed() {
  WebServerClass* webRequest = myWebServer.getRequest();

  // http://xxx.xxx.xxx.xxx/led?val=1
  if (webRequest->hasArg("val")) {
    int value = webRequest->arg("val").toInt();
    digitalWrite(ledPin, value);
  }

  String reply = "LED is now ";
  reply += digitalRead(ledPin) ? "OFF" : "ON";
  webRequest->send(200, "text/plain", reply);
}

void handleTemp() {
  WebServerClass* webRequest = myWebServer.getRequest();

  String reply = "45";
  webRequest->send(200, "text/plain", reply);
}
```

arimukhlas, 2023-07-25:
> 

Oh oke, 
I upload on wemos D1 mini, html can not show value 
but when I upload on my ESP32 Dev Kit V1, it show the value

is there any different config between esp32 and esp8266?

arimukhlas, 2023-07-27:
> > 
> 
> Oh oke, I upload on wemos D1 mini, html can not show value but when I upload on my ESP32 Dev Kit V1, it show the value
> 
> is there any different config between esp32 and esp8266?

Done, I dont know why, but I try flash all and reupload on my wemos, custom html show the value