cotestatnt/esp-fs-webserver #34: Error compiling for board esp8266

arimukhlas, 2023-08-29:
hi @cotestatnt 

I have updated lib to 1.2.4
an error appears when I compile the esp8266 board
```
D:\Documents\Arduino\libraries\esp-fs-webserver-master\src\esp-fs-webserver.cpp: In member function 'bool FSWebServer::begin(const char*)':
D:\Documents\Arduino\libraries\esp-fs-webserver-master\src\esp-fs-webserver.cpp:125:16: error: 'using WebServerClass = using ESP8266WebServer = class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>' {aka 'class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>'} has no member named 'enableDelay'
  125 |     webserver->enableDelay(false);
      |                ^~~~~~~~~~~

.
.
.
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

```

I have also updated the board manager for esp8266, but the error still appears
is there any solution?

BogyS, 2023-09-01:
The method is defined in the ESP32 WebServer, so the call should be moved inside
#ifdef ESP32
...
#ednif

The default value is true and I don't know why it had to be set to false.
In the default case it just add a delay(1) in the handleClient() method if there is no connected client.

cotestatnt, 2023-09-05:
Sorry, i've added the instruction at the wrong line.

New release contains the fix.

closed 2023-09-05 by cotestatnt