cotestatnt/esp-fs-webserver #35: Disable Captive Portal??

IAmOrion, 2023-11-28:
How on earth do I disable the captive portal with this library? I'm using it in AP mode only, but it still presents the captive portal popup when I connect to it (which defaults to /setup) - but I want to disable this, I don't need it at all.

many thanks in advance

cotestatnt, 2023-11-28:
Hi @IAmOrion, unfortunately I have not provided an explicit instruction to deactivate the captive portal.

You need to comment out these lines in the file [esp-fs-webserver.cpp](https://github.com/cotestatnt/esp-fs-webserver/blob/master/src/esp-fs-webserver.cpp#L105)

```
   // Captive Portal redirect
    webserver->on("/redirect", HTTP_GET, std::bind(&FSWebServer::captivePortal, this));
    // Windows
    webserver->on("/connecttest.txt", HTTP_GET, std::bind(&FSWebServer::captivePortal, this));
    // Apple
    webserver->on("/hotspot-detect.html", HTTP_GET, std::bind(&FSWebServer::captivePortal, this));
    // Android
    webserver->on("/generate_204", HTTP_GET, std::bind(&FSWebServer::captivePortal, this));
    webserver->on("/gen_204", HTTP_GET, std::bind(&FSWebServer::captivePortal, this));
```

In any case, if you need to create some kind of HMI, I recommend you take a look at this new version of the library based on the [ESPAsyncWebServer ](https://github.com/cotestatnt/ESPAsyncWebServer) component which, compared to [the one integrated](https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer) into the core, has greater reactivity and better page loadtime.

async-esp-fs-webserver
https://github.com/cotestatnt/async-esp-fs-webserver

IAmOrion, 2023-11-28:
> Hi @IAmOrion, unfortunately I have not provided an explicit instruction to deactivate the captive portal.
>
> You need to comment out these lines in the file [...]

Perfect thanks.  I figured I could comment that stuff out but was hoping I was just missing an easy method/function to disable but you answered that also so that's great, thanks :) 

I shall take look at the newer async library now too, many thanks

closed 2023-11-28 by IAmOrion