cotestatnt/esp-fs-webserver #38: What is the libraries need to use this library?

toneytang, 2023-12-25:
I am very intrested in your projject. But have no idea how to use it. Do you have more detailed help file?

Thanks.

cotestatnt, 2023-12-25:
Hello @toneytang and thank you for your interest.

The only external dependency is [ArduinoJson](https://arduinojson.org/v6/doc/installation/) library, everything else is part of the Arduino core for ESP32 or ESP8266 and therefore you don't need to install anything else.

With [latest release](https://github.com/cotestatnt/esp-fs-webserver/releases/tag/1.2.8) I have deeply revisited the built-in /setup web page: now you can update the firmware OTA and also upload any source files (there are many nice examples included in the "data" subfolder).

In this regard, in the examples I tried to show all the functions that the library makes available and they are certainly the first point to start from.
What would you like to use it for? WiFi manager and parameter configuration or to design a sort of User Interface?

toneytang, 2023-12-26:
Hi, @cotestatnt . Thank you for your reply. I just managed it working under Arduino IDE yesterday. The project is wonderful, it satisfies almost everything of my needs. So, the thing I need to figure out is how to customize the webpage and learn how to use it. 

I am going to use it for an little sensor to store the data on an SD card and also has web ui to view the data easily.

toneytang, 2023-12-26:
By the way, may I ask what is the size for the FS? I am using ESP32-C3-MINI-H4 module.

cotestatnt, 2023-12-26:
> how to customize the webpage 

This a possibility, but not a must: the `/setup` web page can be customized injecting your own HTML/CSS/JS code (check the customHTML.ino example), but it is mainly intended as an endpoint useful for configure a project, not as UI.. 
You could upload a complete custom web sources as for UI, but it'your choice offcourse. I have an old project similar to your needs, if you like I can provide you.

>By the way, may I ask what is the size for the FS? 

This depends from the board and not from MCU type. The SPI flash memory with ESP32 is always external, so every board maker can choose the size he need. Keep in mind that library can work also directly from SD memory aniway because it's indipendent from filesystem type.

closed 2024-01-10 by cotestatnt