cotestatnt/esp-fs-webserver #39: NO ISSUE, just question FILE DOWNLOAD ?

MyRaspberry, 2024-01-07:
play with a ESP32-S3 N16R8 
and setup FFAT 10MB partition.
and write dir and file (append 1 line / min )
**/data/readings.csv**

does **THIS** tool help me to DOWNLOAD that file via Browser?

the example FSBrowser i use, not even print that file (content)
[my status : at GIT](https://github.com/MyRaspberry/ESP32-S3_N16R8_w_Arduino/)

THX for sharing your work

cotestatnt, 2024-01-07:
Hi @MyRaspberry 

Yes, it can.
Some time ago I've written a sketch to visualize a csv, edit and download 
I just have to remember where I saved it.... 

However, it has little to do with the library in question, it's just HTML and Javascript coding.

MyRaspberry, 2024-01-07:
i just worry about the file size can grow over the memory of the chip,
what also limits the html it can produce dynamically,
so how JS could serve big files?

cotestatnt, 2024-01-08:
It depends on the size of your csv file.
Aniway, this library could serve web pages even from a SD memory card if a lot of space is needed.

Naturally, the larger the size of the csv file, the longer it will take to load the page that displays its content.

This is the result with my ESP32-S3 (only 1.3Mbyte for LittleFS filesystem) using my smartphone as Access Point.
Test with a file of approximately 110kb (1000 records) 
[image]

and this instead with a file of approximately 1000kb (8000 records)
[image]

If you only need to download, the problem does not arise because it is not necessary to upload the contents of the file to the web page as in this case.
It will be sufficient to populate a list with the files to download (if there are more than one) and a button to start the download.

MyRaspberry, 2024-01-08:
thanks, 

anyhow i started coding the file thing a ? better ? way,
making daily files like "data/readings_2024-01-08.csv"
will see tomorrow if it works.

and then experiment with your lib

thanks for your time

cotestatnt, 2024-01-08:
> making daily files like "data/readings_2024-01-08.csv"

This is a great idea.
The download page at this point will only have to create a list based on the contents of the filesystem and allow you to download the files.

If you need, I have some working sketches that do similar things (not with csv, but basically it's the same behaviour)

MyRaspberry, 2024-01-09:
-1- that first part seems to work:
find {"type":"file","name":"data/readings_2024-01-09.csv", "size": 21267 B } _( ! size after 8 hours already )_

`,rec,datetimes,A0,A1,A2,
,1785,2024-01-09 00:00:49,0.00,20.05,1.03,
,1789,2024-01-09 00:01:49,0.00,18.95,0.71,
,1793,2024-01-09 00:02:49,0.00,19.10,0.61,
...`

tomorrow i know how big it will be.

-2- i can also now print the content using the **server.streamFile**
`  String contentType = getContentType(path);
  File file = FILESYSTEM.open(path, "r");
  server.streamFile(file, contentType);
  file.close();
`
inside my added function to the FSBrowser code : 
`void handleFilePrint() { // on /print?file=/data/readings.csv`

-3- BUT THAT IS NOT A DOWNLOAD ? HOW ?
i also failed to find something in your 
<b>esp-fs-webserver-master\src\esp-fs-webserver.cpp</b>

-4- i am lost,
while years ago i pay esp2866 and SPIFFS and the /data/ dir in Arduino code and a Arduino IDE TOOLS menu entry UPLOAD
( and after using Circuit Python just come back to Arduino IDE )
now i see that is ?DISCONTINUED?

and having a ESP32-S3 N16R8 module i look into using FFAT
where i miss the way to handle any files there regarding UPLOAD & DOWNLOAD..
so finding FSBrowser i hoped that might be the way, but that is very incomplete.

also i miss to see anything ? regarding other tools
is there something from the Espressif tools?

or ? i have one of that boards with 2 USB C ports,
is there a code ?for Arduino IDE? 
to use one port of that board as a **USB STICK** with file access to FFAT?
HA, that would be great

MyRaspberry, 2024-01-10:
thanks anyhow,
i got the basics working.. see GIT

cotestatnt, 2024-01-10:
Hi @MyRaspberry 

I'm happy you already solved it.
In any case, if you want to take a look, I added an example taking inspiration from your needs

https://github.com/cotestatnt/esp-fs-webserver/tree/master/examples/csvLogger

cotestatnt, 2024-01-10:
Of course you can use it! 
The library is compatible with any type of Espreffif microcontroller including the ESP8266.

If you prefer to use FFat, simply include the "FFat.h" file and replace **LittleFS** with **FFat** wherever you see it in the sketch.

As for uploading the data folder, it is managed directly by the library using the included `/setup` page,  **Update & FS** tab.

[image]

Once you have uploaded your sources, you can edit them directly using the included `/edit `page
Soon I will add a new version of `/edit` that allows you to download files, but in the meantime you can simply copy and paste to transfer any changes to your PC

Regarding your working HTML code, you can simply put it on a specific page: if you don't want write from beginning, open dev tools of your browser, move to the "source" tab and copy the content.
If the name page is `index.htm` or `index.html` it will be used as default page

cotestatnt, 2024-01-10:
The print screen is from the [twin library](https://github.com/cotestatnt/async-esp-fs-webserver) that uses the ESPAsyncWebServer library for the webserver instead of the one included in the core, but the result does not change (I develop them more or less in parallel)

From some tests I've done it seems to be faster and more responsive.

MyRaspberry, 2024-01-11:
as expected nothing is easy....
first ( not yet reading above and with my possibly wrong thinking ) i try a extra working environment

- on a laptop ( win 10 )
- Arduino IDE 1.8.19
- installed lib esp-fs-webserver 1.2.8 
- add download copy example from MASTER
- board ESP32 : **NODE ESP-32S V1.1** 
- press the RIGHT BUTTON for UPLOAD this sketch 
( for first time i used erase all flash [enabled], who knows what i play last year ?CircuitPython? )

try set laptop wifi to 
**ESP_AP**
password 123456789
from laptop
get:

> Oops.
> 
> The site at http://www.msftconnecttest.com/redirect has experienced a network protocol violation that cannot be repaired.
> 
> The page you are trying to view cannot be shown because an error in the data transmission was detected.
> 
>     Please contact the website owners to inform them of this problem.
>

cotestatnt, 2024-01-11:
I'm sorry @MyRaspberry, but a pull request I've merged some times ago has introduced this bug due to a wrong reply when ESP is acting as captive portal ( `"https:://" ` instead `http://`).

Once an ESP32 has succesfull connected to a SSID it store the credentials, so I've never realized before about this bug.

With last commit and releases I've fixed it. 
I've added also some more features like manual WiFi configuration (fixed IP, gatway, subnet)

MyRaspberry, 2024-01-11:
sorry, had to fail with a other idea first.....

// 12.1.2024 update to v 1.3.0 and take also that new example

Terminal show continuous booting
```
05:45:39.592 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
05:45:39.639 -> configsip: 0, SPIWP:0xee
05:45:39.639 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
05:45:39.639 -> mode:DIO, clock div:1
05:45:39.639 -> load:0x3fff0030,len:1184
05:45:39.639 -> load:0x40078000,len:13260
05:45:39.639 -> load:0x40080400,len:3028
05:45:39.639 -> entry 0x400805e4
05:45:39.780 -> ERROR on mounting filesystem. It will be formmatted!
05:45:39.780 -> ets Jun  8 2016 00:22:57
05:45:39.780 -> 
05:45:39.780 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
05:45:39.780 -> configsip: 0, SPIWP:0xee
05:45:39.827 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
05:45:39.827 -> mode:DIO, clock div:1
05:45:39.827 -> load:0x3fff0030,len:1184
05:45:39.827 -> load:0x40078000,len:13260
05:45:39.827 -> load:0x40080400,len:3028
05:45:39.827 -> entry 0x400805e4
05:45:48.649 -> ets Jun  8 2016 00:22:57
05:45:48.649 -> 
```
again, above i mentioned that i use partition DEFAULT ( whatever that means )
[image]

cotestatnt, 2024-01-12:
You are trying to use a FFat based sketch selecting a SPIFFS based partition scheme (the default one),
Of course it will not work!

Also, up until now you've always said you have an ESP32-S3, but the board you selected uses a "normal" ESP32 so I'm guessing this is a different board?
In this case, select a variant that also has the FFat partition among the choices, or select a generic ESP32 dev-module so you have all the options available.

A curiosity... but why are you so sticky to the FFat partition? If you left LittleFS, the sketch would work as it was without having to specify the partition scheme every time

MyRaspberry, 2024-01-12:
oh sorry,
-1- my idea was to test a new ESP32-S3 N16R8
with Arduino, so installed Arduino IDE 2.2.1 to learn that too.

-2- my first impression was that i can not use that N16R8
so i dig deeper and that leaded to using FFAT

as i am working and data-collecting to N16R8  
in that bigger evaluation project already
[status](http://kll.byethost7.com/kllfusion01/downloads/ESP32_FSBrowser_R16N8.png)

-3- as described above, i did a separate test setup for your lib.
-a- laptop
-b- Arduino IDE 1.8.19
-c- a NODEMCU ESP32-32s v1.1
as i got the impression you need that Arduino /data/ dir thing
where i read that not work in the new IDE?

again, just tell me what you want me to do with your last code,
to test without change?
- board
- IDE
- tools board settings

pls, i want to study that.

cotestatnt, 2024-01-12:
> as i got the impression you need that Arduino /data/ dir thing where i read that not work in the new IDE?

The FS upload plugin for IDE 1.8.x (which don't work with IDE 2.x) is no more necessary because you can upload your files through the esp-fs-webserver library's included web interface.

> second try used 192.168.1.213 | 192.168.1.1 | 192.168.1.1

I don't think the [subnet mask](https://avinetworks.com/glossary/subnet-mask/) you used are a valid value. Usually I set as 255.255.255.0

If you edit manual IP or other parameters for no-DHCP configuration, you have to save configuration before start the new connection, otherwise on next restart it will not be used.

MyRaspberry, 2024-01-13:
> 255.255.255.0
my bad

MyRaspberry, 2024-01-13:
after some looks i got to say:

## -a- 
this project needs a FAVICON.ICO
[ like ](http://kll.byethost7.com/kllfusion01/downloads/ESP32_fs_webserver.png)
with change in :
**/data/index.htm**
<link rel="icon" type="image/x-icon" href="assets/favicon.ico">
and here my idea, pls you use mine if you like it, even i am not a real artist.
pls copy from :
[ here ](https://github.com/MyRaspberry/ESP32-S3_N16R8_w_Arduino/tree/main/ESP32_Data_Collect_FFAT_FSBrowser/code/data)

## -b- 
i miss a ?maintenance? view about the 'LittleFS' DRIVE in PSFLASH 
* space? total free used
* dir list ( not only the .CSV like in CSV work HOME page file sector )

## -c-
this project `csvLogger` : funtion is impressive,
but the code is 3 level over my head to understand
if you look at my GIT you see i just managed to learn .h .cpp class 
and have html download button browse select one file, but the way seems to be same?

server ON /edit [POST] >> handleFileUpload() >> server.upload()

all just hidden in your complex structure.
but i not see where you scan the PC selected `/data/` dir and upload ALL files?
is it a ONE by ONE via a JS loop?

## -d- 
anyway 
* we both got the `poor` FSBrowser to a FULLER USE
* and i am happy that using `N16R8` via Arduino IDE is tricky but possible.

- - took me a week to find that out, funny that `google` could not help me with that 
possibly my bad english. 
thanks and greeting to Rome from Chiang Mai

cotestatnt, 2024-01-13:
I'll try to respond to your comments

- A 
favicon.ico is something linked to the final application chosen by the user, so inserting it at library level seems useless to me. Of course I could put a default one to customize if necessary... I'll think about it.
- B
the methods for retrieving this information is tied to the choosen filesystem. In the twin library [async-esp-fs-webserver](https://github.com/cotestatnt/async-esp-fs-webserver), is possible to assign a callback function for this task. I will include this possibility in this library soon
- C
yes, it's all done with a [Javascript function ](https://github.com/cotestatnt/esp-fs-webserver/blob/master/setup-ui/data/app.js#L552) and with [HTML input file](https://github.com/cotestatnt/esp-fs-webserver/blob/master/setup-ui/data/setup.htm#L127) element with id=listing
- D
ok, I'll do some checking
- E
Thank you from Rome for the valuable feedback

MyRaspberry, 2024-01-13:
now i try 
* Arduino IDE 2.2.1 ( after NODEMCU ESP32 S OK )
* board ESP32-S3 N16R8
* erase all FLASH ENABLED
*
?? WHAT NOW??

* FLASH size 4MB
* Partition 4MB 1.5MB SPIFFS

* mobile wifi >> ESP_AP
* *                password 123456789
* * Authenticating..
* * Couldn't connect Try again

-1- pls help with the tools board settings for ESP32-S3
-2- why now login problems?

cotestatnt, 2024-01-13:
I'm sorry something must have gone wrong with the latest commits...
for now I would suggest starting the connection manually with the classic `WiFi.begin()` and avoiding calling the library `startWifi()` method

I'll fix this bug and then I'll make a new release.

MyRaspberry, 2024-01-14:
ok, n.p. can wait,
just wanted to confirm if the rest worked and run pure FFat tools over it

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4

================================
Chip Model: ESP32-S3
Chip Revision: 0
with 2 core
Flash Chip Size : 16777216 
Flash Chip Speed : 80000000 
Total Heap : 397212 
Free Heap  : 372320 
PSRAM Total :  8386279, Free : 8386019, Used : 260

Features included:
 
 2.4GHz WiFi
 Bluetooth LE
 
 

FFAT Total :   10240000, Free :   10231808, Used :       8192
Listing directory: /
  DIR : setup
Listing directory: /setup
  FILE: config.json	SIZE: 17

```

MyRaspberry, 2024-01-16:
update:
[try v 1.3.2](http://kll.byethost7.com/kllfusion01/infusions/articles/articles.php?article_id=234)

cotestatnt, 2024-01-16:
Hi @MyRaspberry, thanks for your review!
I just published a new major release (due to the big and deep changes).

If you want test also this release, please keep in mind that some small changes are needed on on pre-existing sketches.
In any case, the advice is always to start with the examples included.

MyRaspberry, 2024-01-16:
[try v 2.0.0](http://kll.byethost7.com/kllfusion01/infusions/articles/articles.php?article_id=234#example5)

cotestatnt, 2024-01-19:
Hi @MyRaspberry 
If you like, try the latest release.
I've fixed some of the observations you made such as wrong printing files order.
As regards the "Google" IP in access mode, it seems that with this trick you can make the captive portal work even on the most finicky devices (for example some Samsungs smartphones)

Furthermore, in the [csvLogger.ino](https://github.com/cotestatnt/esp-fs-webserver/blob/master/examples/csvLogger/csvLogger.ino#L18), I have included an example of how it is possible to insert a custom favicon encoded in base64.

The `/edit` web page must be explicitly enabled. However I have adjusted all the examples to include it (and possibly comment the related instruction)

MyRaspberry, 2024-01-20:
[ test 9 and 10 ](http://kll.byethost7.com/kllfusion01/infusions/articles/articles.php?article_id=234&i=1#example6)