cotestatnt/esp-fs-webserver #40: Problem with ActionBox rename

AlainF-ing, 2024-01-07:
I am using arduino-cli   and ESP32-S3 D1 Mini
I was unable to change Options Box Label in your example CustomHTML.ino.
Even if I use the    #define CLEAR_OTIONS 1     the label did not changed.
I have to get rid of the config.json (actually I rename it) in order to see my new labels.
In config.json I have noticed that options labels seems to be saved EX:"param-box1": "Mes Options".
Exception are "Setup" and "Update&FS"
This is confusing.

cotestatnt, 2024-01-08:
Hi @AlainF-ing and thanks for your interest!
The behavior you encountered is normal: in order not to unnecessarily overwrite the file every time you restart, when the key is present the file is not touched.
This unfortunately has the drawback you encountered, but i prefer this way because in the end, it can only cause problems during the development phase of the project

Regarding the `#define CLEAR_OTIONS`, this is used only in the example customHTML.ino and not in customOptions.ino

If you look at the code, you can find this:

```
#if CLEAR_OPTIONS
  if (myWebServer.clearOptions())
    ESP.restart();
#endif
```
So, when you need to clear the config file the method to be used is `clearOptions()`, but of course if you use the `/edit` web page the result is the same.

AlainF-ing, 2024-01-08:
Ok and thanks for your quick answer.
As a comment, I am a fan of the FSBrowser.ino on Little FS for ESP8266. For ESP32 this was a disappointment until I
found your library. 
Very good work.

cotestatnt, 2024-01-08:
Thanks for your comment!

If you like this library, I would recommend you also to [take a look at this version](https://github.com/cotestatnt/async-esp-fs-webserver) which, instead of the WebServer core library, is based on the ESPAsyncWebServer library.
ESPAsyncWebServer library and its dependencies are included locally in the sources, so you don't need to install anything else.

`async-esp-fs-webserver` and `esp-fs-webserver` can coexist without any problem or interference between them.

Even though I am trying to develop the two projects in parallel, I won't deny that there is more effort on the second one because it has better performance in terms of speed and responsiveness of the pages.

closed 2024-01-17 by cotestatnt