cotestatnt/esp-fs-webserver #41: Crash loading examples

ktiedt, 2024-01-22:
I've tried several of the examples and they all seem to crash and reboot...  I've tried with and without config.json files, empty file, just an empty json object... all result in the same failure.

```
Hard resetting via RTS pin...
--- Terminal on /dev/cu.wchusbserial10 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

/
|__ favicon.ico         (1150 bytes)
|__ index.htm   (2004 bytes)

/css
|__ pico.fluid.classless.css.gz         (5387 bytes)

/img
|__ espressif.jpg       (34305 bytes)

/setup
|__ config.json         (0 bytes)

[E][.pio/libdeps/esp32dev/esp-fs-webserver/src/esp-fs-webserver.cpp:228] startWiFi():   Failed to deserialize file, may be corrupted
 EmptyInput

[I][.pio/libdeps/esp32dev/esp-fs-webserver/src/esp-fs-webserver.cpp:208] startAP():     AP mode.
Server IP address: 8.8.8.8Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dad5b  PS      : 0x00060130  A0      : 0x800dc2d5  A1      : 0x3ffb2110  
A2      : 0x00000038  A3      : 0x3ffb2148  A4      : 0x3f4012b1  A5      : 0x3f41161f  
A6      : 0x00000000  A7      : 0x3f400317  A8      : 0x80167520  A9      : 0x3ffb2100  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0xffffbfff  A15     : 0x3f4012b1  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000040  LBEG    : 0x400e8ccc  LEND    : 0x400e8d12  LCOUNT  : 0x0000003c  

Backtrace: 0x400dad58:0x3ffb2110 0x400dc2d2:0x3ffb2140 0x400dc3a3:0x3ffb2190 0x400d23e6:0x3ffb2230 0x400df0de:0x3ffb2290

ELF file SHA256: 89734c2ea7361308

Rebooting...
```

cotestatnt, 2024-01-22:
Hi @ktiedt 

The problem is due to the wrong size of `/setup/config.json` file.
I assume you have tried the `simpleServer.ino` example which unfortunately due to a bug creates an invalid configuration file.

Do an "erase flash" to delete it and try one of the examples that provides custom options (e.g. customOptions.ino)
Or simply add a "pretend option", for example with statement called before `begin()`

`myWebServer.addOptionBox("Test");`

This way the file is created correctly.
In the meantime I will fix this bug for the next release.

AlainF-ing, 2024-01-22:
I had a similar problem on my ESP32-S3.à
 In my case the corruption the WiFi stored parameter (I guess ).
 I have tried to load with flash erase but it did not work.  
I ended up using a simple program with ssid and password for my WiFi. 
This time the S3 came back.
I can not explained what happened , even debug prints in setup() were not executed !

closed 2024-02-17 by cotestatnt