cotestatnt/esp-fs-webserver #43: on version 2.0.2 ,the saveOption function does not take effect when used in loop.

Zhantongoscar, 2024-02-07:
the function:  myWebServer.addOption(LONG_LABEL, longVar); could save data to fs and could be loaded later, but when i use saveOptions();
it work as follow code :   // Config file will be opened on the first time we call this method
..
  myWebServer.saveOptionValue(LONG_LABEL, longVar);
..
  Serial.println(F("Application options saved."));
It seem not take effectect, is there somthing wrong in code:?
my code is as follow:
void loop() {
  myWebServer.run();

  if (! digitalRead(BTN_SAVE)) {

// 
    loadOptions();
    Serial.printf("Application option loaded after web requestlongVar =%d\n",longVar);
    longVar++;
    Serial.printf("finished add n option loaded after web requestlongVar =%d\n",longVar);

  // 
  saveOptions();
   Serial.printf("---------------------\n");

// 
    loadOptions();
    Serial.printf("Application option loaded after web requestlongVar =%d\n",longVar);
    longVar++;
    Serial.printf("finished add n option loaded after web requestlongVar =%d\n",longVar);

  //把longVar 保存的config文件中
  saveOptions();
     Serial.printf("========================\n");
    }

  delay(1000);

}

===========
Serial output：
This are the current values stored: 

LED pin value: 2
Bool value: true
Long value: 1010
Float value: 15.500
String value: Test option String
Dropdown selected value: Monday

Application option loaded after web requestlongVar =1010
finished add n option loaded after web requestlongVar =1011
Application options saved.
---------------------

This are the current values stored: 

LED pin value: 2
Bool value: true
Long value: 1010
Float value: 15.500
String value: Test option String
Dropdown selected value: Monday

Application option loaded after web requestlongVar =1010
finished add n option loaded after web requestlongVar =1011
Application options saved.
========================

cotestatnt, 2024-02-17:
Hi @Zhantongoscar.
Sorry for the late reply and thanks for reporting the bug.
I'll get to work immediately to understand what could cause this malfunction

cotestatnt, 2024-02-17:
Hi again.

The error was in the **customOptions.ino** example.
 The method for saving options must be called with a true parameter in order to overwrite the `config.json` file
```
myWebServer.closeConfiguration(true);
```

Zhantongoscar, 2024-02-18:
thank cotestatnt for kindly answer! the esp_fs_weber lib is a great lib i have ever used in arduino. I follow it for years and use it as my base application in iot. thanks again!

Zhantongoscar, 2024-02-18:
by the way, we should update the code "myWebServer.closeConfiguration(true);" in save function at  customOptions.ino example.

closed 2024-02-18 by Zhantongoscar