cotestatnt/esp-fs-webserver #44: Connect to WIFI that doesn't require a password

Nickhofr, 2024-02-18:
Hi our community WIFI network does not require a password. Yet I am required to input a password when I try and connect to Wifi.

cotestatnt, 2024-02-19:
Hi @Nickhofr 
The function for connecting to WiFi is integrated into the setup page and is defined here.

To do what you need you could simply make your own custom copy without verifying the password field and associate your function with the connection button by removing the default one.

For example:
- add this literal string to you project (it's simply the copy and paste of the function without checking the password) and at the bottom an instruction to change the association with the click of the button
```
function doConnectionCustom(e, f) {
  console.log('This is password-less WiFi connection function');

  if ($('ssid').value === ''){
    openModalMessage('Connect to WiFi','Please insert a SSID and a Password');
    return;
  }
  var formdata = new FormData();
  formdata.append("ssid", $('ssid').value);
  formdata.append("password", $('password').value);
  formdata.append("persistent", $('persistent').checked);
  if (typeof f !== 'undefined')
    formdata.append("newSSID", true);
  if ($('no-dhcp').checked) {
    formdata.append("ip_address", $('ip').value);
    formdata.append("gateway", $('gateway').value);
    formdata.append("subnet", $('subnet').value);
  }
  var requestOptions = {
    method: 'POST',
    body: formdata,
    redirect: 'follow'
  };

  $('loader').classList.remove('hide');
  var s;
  fetch('/connect', requestOptions)
  .then(function(res) {
    s = res.status;
    return res.text();
  })
  .then(function(data) {
    if (s === 200) {
      if (data.includes("already")) {
        openModalMessage('Connect to WiFi', data, () => {doConnection(e, true)});
        $('loader').classList.add('hide');
      }
      else
        openModalMessage('Connect to WiFi', data, restartESP);
    }
    else
      openModalMessage('Error!', data);
    $('loader').classList.add('hide');
  })
    
  .catch((error) => {
    openModalMessage('Connect to WiFi', error);
    $('loader').classList.add('hide');
  });
}

$('connect-wifi').removeEventListener("click", doConnection);
$('connect-wifi').addEventListener("click", doConnectionCustom);
```
- add this statement to your webserver before starting it

```
myWebServer.addJavascript(custom_wifi_conenction, "wifi");
```

Nickhofr, 2024-02-19:
Hi thank you I appreciate your help. Forgive me but this is a little new for me. Could you give me more of a step by step instructions. Which file to edit, where to put code? I see there are a number of files in the setup folder.  TIA.

Nickhofr, 2024-02-19:
My persistence has payed off. 
I was able to copy your code of the function doConnection and replace the code from the app.js file. 
 I followed the instructions from the readme.md file to run node minify.js in cmd.
 The issue that took me a while to resolve was that the compiler made a "static const unsigned char SETUP_HTML[] PROGMEM =" in the setup_htm.h rather then "static const unsigned char _acsetup_min_htm[] =". 
That resulted in an error when I tried sending the program to the esp32 in the arduino program. 
Eventually I renamed the heading in the setup_htm.h. It worked. Connected to WIFI without having to issue a Password. 
Huge thanks!

Nickhofr, 2024-02-20:
I would say my issue is now it won't store my SSID. Every time I restart my Esp32 It sets up an AP rather then connect to the last WIFI SSID that I connected to.

Nickhofr, 2024-02-21:
Hi can you help me this? I have tried a few things and can't get it to store my Wifi SSID.

closed 2024-02-22 by Nickhofr