cotestatnt/esp-fs-webserver #45: Won't Store Wifi Credentials when Password is empty.

Nickhofr, 2024-02-22:
I figured I will open up a new issue with this. I was able to alter the code with your help so that the esp32 would connect to WIFI SSID that doesn't require a password. It worked perfectly. The issue now is it won't save that SSID or connect to it when I reset the esp32. Instead it sets up an AP right away after reset, as if there is no WIFI SSID stored that it can connect to.

cotestatnt, 2024-03-07:
Hi @Nickhofr and sorry for late reply.

Last commit should fix this bug. I will do some more tests and then I will draft a new release.

sjmf, 2024-03-07:
Hi @cotestatnt  - just to note your commit b176a1c also wiped out the changes from PR #46. Perhaps you force-pushed?

cotestatnt, 2024-03-08:
Hi @sjmf 
I'm sorry. I've used Github Desktop from my second laptop and I sure made some mess.
I've just restored your PR and now it should be OK.

Nickhofr, 2024-03-08:
Works Great! Thank you so much!

closed 2024-03-09 by cotestatnt