cotestatnt/esp-fs-webserver #53: Compilation error

HaViGit, 2024-07-30:
First of all thanks for sharing this code. Unfortunately I am facing a problem while compiling for an ESP32-S3 board.  The error message below is shown:

Compilation error: no matching function for call to 'FSWebServer::FSWebServer(fs::LittleFSFS&, WebServerClass&)'

Do you know how I can solve this?

cotestatnt, 2024-07-30:
What version of the ESP32 core for Arduino are you using?
I tried to compile the examples with the latest one available (v3.0.3) and I don't encounter any problems.

I think you inserted an obsolete link in the IDE because unfortunately there are too many outdated tutorials circulating online.

[The correct json link ](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) to be used can be found in the official Espressif documentation

HaViGit, 2024-07-30:
Thanks for the super fast response! I'm using the CrowPanel below and trying to compile a sketch for a Google Calendar (see below), can you see if that's the problem?

[https://www.elecrow.com/wiki/esp32-display-702727-intelligent-touch-screen-wi-fi26ble-800480-hmi-display.html](url)
[https://github.com/cotestatnt/Arduino-Google-API/tree/main/examples/GoogleCalendar](url)

cotestatnt, 2024-07-30:
The development of the Arduino-Google-API library no longer continued due to lack of interest.
The latest release of the Arduino core introduced many changes and I don't know if the library still works correctly.

If I can find some time to spend on it, I'll try to fill out the Google Calendar example.

HaViGit, 2024-07-30:
That clears up a lot, thanks for your help.

closed 2024-11-29 by cotestatnt