cotestatnt/esp-fs-webserver #55: Can show the time and set time by web?

zerotoney, 2024-11-28:
Hi, 

Can I show the ESP32 time on web and change the time by time_picker? 

If yes, can you please help?

Thank you. 

Toney

cotestatnt, 2024-11-29:
In the included examples, you can find [withWebSocket.ino](https://github.com/cotestatnt/esp-fs-webserver/blob/master/examples/withWebSocket/withWebSocket.ino).
In this example, a websocket connection is used to update the time shown on the page.

The ESP32, after synchronizing with the NTP server, sends the updated time (in epoch time format) to the HTML page once per second using the websocket channel.

The included example page, updates the displayed time accordingly (the Javascript [`function parseMessage(msg)`](https://github.com/cotestatnt/esp-fs-webserver/blob/master/examples/withWebSocket/data/index.htm#L68C7-L68C35))

In order to achieve your goal, you can exploit this example by adding a Javascript function on the HTML side that sends the selected time with the datetime picker and the related handler on the ESP32 side.
Assuming you are working without a NTP server (otherwise why would you need to set the time from the web page?):

### HTML / Javascript
```
HTML section
...... 
<div>
  <input type="datetime-local" id="datetimePicker">
</div>
.....

// Javascript section
// Function to handle the 'change' event
function onDateTimeChange(event) {
  var datetimePicker = event.target; // Get the element that triggered the event

  if (datetimePicker.value) {
    var selectedDateTime = new Date(datetimePicker.value); // Convert the value to a Date object
    var epochTime = Math.floor(selectedDateTime.getTime() / 1000); // Convert to epoch time in seconds
    console.log('Epoch Time:', epochTime); // Log the epoch time

    // Construct the URL for the GET request
    var url = '/setTime?timestamp=' + epochTime;

    // Send the GET request
    fetch(url, { method: 'GET' })
      .then(response => {
          return response.text(); // Parse the response as text
      })
      .then(data => {
          console.log('Server Response:', data); // Log the server's response
      })
      .catch(error => {
          console.error('Error while sending:', error); // Log any errors
      });
  } 
  else {
    console.log('No date selected.'); // Log a message if no date is selected
  }
}

// Attach the 'change' event listener to the datetime picker
document.getElementById('datetimePicker').addEventListener('change', onDateTimeChange);
```

### ESP32 firmware

```
// HTTP request handler
void handleNewTime() {
  if(myWebServer.hasArg("timestamp")) {
    uint32_t epochTime = myWebServer.arg("timestamp").toInt();
    struct timeval tv;
    tv.tv_sec = epochTime;  // Seconds since epoch
    tv.tv_usec = 0;         // Microseconds (not necessary)
    settimeofday(&tv, NULL);    
    myWebServer.send(200, "text/plain", "OK");
  }
  else {
    myWebServer.send(500, "text/plain", "Bad request");
  }
}

void setup() {
....
....
  // Add set time  handlers
  myWebServer.on("/setTime", HTTP_GET, handleNewTime);
....

```

zerotoney, 2024-11-30:
Thank you so much!

It works well.

closed 2024-11-30 by zerotoney