cotestatnt/esp-fs-webserver #56: remove verification page

zerotoney, 2024-12-04:
Hi, It's me again.

How to remove the verification page?

Each time when I connect the AP, it will jump to the verification page before working properly. 

How can I remove that feature?

Thanks. 

Toney

zerotoney, 2024-12-04:
[image]

cotestatnt, 2024-12-22:
Hi @zerotoney 
The redirect is handled starting from [here](https://github.com/cotestatnt/esp-fs-webserver/blob/master/src/esp-fs-webserver.cpp#L97)

zerotoney, 2024-12-29:
oh, That is great!! Thanks a lot.

BTW, can I switch this function off when I using this library? I am not confident if I modify the library files. 

^_^.

zerotoney, 2025-01-06:
[image]
Finanlly, I comment them out. :)

closed 2025-12-07 by cotestatnt